pimoroni/st7789-python #3: Please Help!

Zenot1cc, 2020-10-10:
Sorry if im being dumb, but im trying to use this with a RP, i have installed installed "sudo pip install st7789" and still have no image, i assume there is something else i need to install, but the instructions were not clear. Please Help!

Cipher0009, 2020-10-13:
Yeah I was having some difficulty with the instructions too, but luckily after a bit of google/debugging I found a solution. You need to make sure that you have enabled SPI in the configuration interfacing options, it should be under option 5 then option p4 when you run "sudo raspi-config".

Zenot1cc, 2020-10-13:
hmm, i have done that, and restarted, and it still doesnt work: the backlight works but nothing else: Cannot think of anything else so any help is appreciated lol

Cipher0009, 2020-10-13:
By chance have you tried running pip install setuptools and then going into the st7789 directory and running setup.py. Are you using raspbian like or the desktop version, and are you using a pirateaudio device?

Zenot1cc, 2020-10-13:
Im using an OS built on Raspbian Desktop, called MoBro. I cannot find the ST7789 Directory for the life of me, either. It should have the exact same file structure as its just an overlay with some extra functionality built on top of it. Where would i be able to find this directory? I only have access to command line aswell as i can only SSH into the Pi

Cipher0009, 2020-10-13:
No worries, I'm connecting to a headless pi zero running raspbian lite as well. If its not too much trouble, could you clone the repo directly and test from there, also if you could share your boot/config.txt maybe its something in there. This was the source I used to get it working 

https://github.com/pimoroni/st7789-python.git

Zenot1cc, 2020-10-13:
hm, there doesnt seem to be a setup.py in the st7789 directory, strange!
The only files are __init__.py and __init__.pyc

I used the same source, and what do you mean by boot/config? 
Im assuming you mean the config file in the SD card when i put it into a computer:[https://mega.nz/file/2jpmBCra#TwXHonFzDhS7b_Tgt1Rp12NPdPil2_9fST2zvmMdy2A](url)

Cipher0009, 2020-10-13:
Where did you run your git clone? Because those are the python files, and sadly the mega upload does not show. Luckily, I am actually about to restart my process on a few other pi zero's because I intended to create a whole home audio system from these devices, so I can just write up my process as a bash script and share it with you.

Cipher0009, 2020-10-14:
Sorry it took me so long, I stepped out for a bit. I was able to load everything I have completed so far to a git-hub repository you are more than welcome to take a look at it, if maybe it is of any help. 

https://github.com/Cipher0009/PiZeroSetupFiles

Zenot1cc, 2020-10-14:
Thank you very much! you're a godsend! However:
when i try to run python gif.py deployrainbows.gif, it gives the error:

Traceback (most recent call last):
  File "gif.py", line 45, in <module>
    spi_speed_hz=80 * 1000 * 1000
  File "/home/modbros/.local/lib/python2.7/site-packages/ST7789/__init__.py", line 124, in __init__
    self._spi = spidev.SpiDev(port, cs)
IOError: [Errno 13] Permission denied

same as when i do the same for image.py
any ideas?

Cipher0009, 2020-10-14:
Anytime! And actually yes thats the exact same error I was initially getting. You said you manually enabled SPI in raspi-config right?

Cipher0009, 2020-10-14:
If so I have a dumb idea, which might work. Try running the python command as sudo, I know thats not typical but in your error message it looks like it failing to initialize because it cant traverse that particular directory. 

And if that doesn't work try appending these lines to your boot/config.txt 

dtoverlay=hifiberry-dac
gpio=25=op, dh

Zenot1cc, 2020-10-16:
Thank you! Sorry for the late reply, i had IRL things to attend to. Using Sudo did indeed work, but no display on the screen, please confirm you are using the same wiring that i am: [https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi.stack.imgur.com%2FbIASO.jpg&f=1&nofb=1](url) 
The backlight turns on, but nothing is displayed

Zenot1cc, 2020-10-16:
@Gadgetoid im very sorry,  i have no idea with Linux, im new to all of this lol, please explain where i can find this?
Im sorry for being clueless, im trying my best!

Gadgetoid, 2020-10-16:
@Zenot1cc if you're just wiring the breakout up and running these examples, you're going to have a bad time. You need to ensure you change the code to match your chosen pinout. And even then it's not guaranteed to work.

Here's where it is in `scrolling-text.py` for example:

https://github.com/pimoroni/st7789-python/blob/fa6195dcb7d12e3969f047ce2b71b2a5b3ee417c/examples/scrolling-text.py#L12-L18

Are you following any kind of guide to set up this display?

Cipher0009, 2020-10-16:
@Zenot1cc I am a bit confused as to what you mean by wiring, are you utilizing a different device? My configurations are with a pi-zero, pi-zero w, and pi 4; in terms of my pimoroni pirate audio devices I am using the line-out device and and 3w dac. And don't worry about being new to linux at all, like thats what linux is all about, learning.

Cipher0009, 2020-10-16:
@Zenot1cc is there any particular reason you are using the Modbros raspbian variant?

Gadgetoid, 2020-10-16:
@Cipher0009 if you solve the riddle to get the posted broken URL working, it'll resolve to this image:

[image]

Which leads me to believe it's a generic ST7789 and not one of our products.

That said, when I write a library called "st7789" I do so with the expectation that someone might want to use it for... an st7789 😆  

(Thanks for jumping in here, by the way!)

Cipher0009, 2020-10-16:
@Gadgetoid Might I be able to ask you a somewhat similar question here, or would it be better if I just open a different issue?

Zenot1cc, 2020-10-16:
@Cipher0009 Im using MoBro as i intend to use this display to show my CPU Temps, but their display adapters don't work for this display, It seemed like a fairly simple case mod i could do without spending out a bunch as i'm a typical college student with fuck all cash.
The only guides ive been following is the photo you shared above, and used a generic ST7789 display from AliExpress
In terms of using a different Device: Im using a Pi Zero W, and a ST7789 display that can be seen in the picture above (I realise the link i sent doesnt work, as im a dumbass, as always)

Cipher0009, 2020-10-16:
@Zenot1cc I see, you are essentially attempting to build your own ibuypower snow-blind element case. Luckily, it just so happens that I was thinking of doing something similar a while back; I mod my computers quite a bit too, even though I'm a broke college student as well, hahaha. I usually just take broken things and rework them.

In your particular case, I would recommend actually just finding an old tv taking that apart, while keeping the logic board w input connections and display together, ofc being insanely careful with the connector cable sometimes its a ribbon cable and that's delicate. Attaching it the display to some type of acrylic to stand it within your case, and then just 3d-printing or finding an empty plastic box for the logic board, to be stored within your case. 

I would recommend going about it this way if your PC usage is not resource intensive. Like I have a 2070 super, but Im not really a gamer so it worked out fine for me. I found this way best for my use case because inserting a second device like the raspberry pi into my pc will either draw more bandwidth from my network, which is not bueno because i need all 1000mbps, or increase my thermals bc second device=more heat, also no bueno.

Cipher0009, 2020-10-16:
It would look like this:
[image]

Cipher0009, 2020-10-16:
@Gadgetoid Might I be able to bother you with a configuration question?

Zenot1cc, 2020-10-16:
@Cipher0009 Thanks bro! I decided to go this route as i have a DeepCool Matrexx 30 that has a small hole next to the PSU, almost the exact right size for a PI Zero!
[image]
I wanted to avoid using a second display, as it could mean i could lose my mouse onto it when im playing:
I typically use my pc for gaming: Running a 1660 Super,  so any resources I can save, i will

Cipher0009, 2020-10-17:
In that case you might be able to pull it off by parsing data from an output log, maybe from third party software like CPU-Z or hwinfo but of course a direct connection might require something like a serial cable. If not the only other way which comes to mind of the top of my head is creating a custom script to export, retrieve, and render your data over wifi or bluetooth. I found some information online which might be of help to you.

https://linustechtips.com/main/topic/1137767-arduino-cpu-temp-monitor-on-16x2-lcd/?do=findComment&comment=13155775

Zenot1cc, 2020-10-17:
Interesting, and thats essentially what MoBro does, without me having to write all the code lol
[https://www.mod-bros.com/en/projects/mobro](url)
Im still in my search to get this working, however, as ive started this project and must finish it or ill kick myself
ill try get that code for @Gadgetoid aswell, ill post that in a few

Zenot1cc, 2020-10-17:
```
disp = ST7789.ST7789(
    port=0,
    cs=ST7789.BG_SPI_CS_FRONT,  # BG_SPI_CSB_BACK or BG_SPI_CS_FRONT
    dc=9,
    backlight=19,               # 18 for back BG slot, 19 for front BG slot.
    spi_speed_hz=80 * 1000 * 1000
)

# Initialize display.
disp.begin()

WIDTH = disp.width
HEIGHT = disp.height

img = Image.new('RGB', (WIDTH, HEIGHT), color=(0, 0, 0))

draw = ImageDraw.Draw(img)

font = ImageFont.truetype("/usr/share/fonts/truetype/dejavu/DejaVuSans-Bold.ttf", 30)

size_x, size_y = draw.textsize(MESSAGE, font)

text_x = disp.width
text_y = (80 - size_y) // 2

t_start = time.time()

while True:
    x = (time.time() - t_start) * 100
    x %= (size_x + disp.width)
    draw.rectangle((0, 0, disp.width, 80), (0, 0, 0))
    draw.text((int(text_x - x), text_y), MESSAGE, font=font, fill=(255, 255, 255))
    disp.display(img)
```
I am using the exact wires that are shown in this picture: 
[image]

Cipher0009, 2020-10-19:
@Gadgetoid Could you please explain the logic flow within ST7789? Essentially, what I am attempting to do is turn my pirate-audio device into a local audio player device/watch. I have been playing with the python code a bit and can render my own images on the screen, but the audio cuts which is definitely because I am not comprehending the logic.

Gadgetoid, 2020-10-24:
@Cipher0009 I don't have time for an in-depth explanation I'm afraid, but maybe my pirate-mp3 example will get you closer to where you're heading? https://github.com/gadgetoid/pirate-mp3

closed 2020-10-24 by Gadgetoid

riotgibbon, 2020-11-13:
> Yeah I was having some difficulty with the instructions too, but luckily after a bit of google/debugging I found a solution. You need to make sure that you have enabled SPI in the configuration interfacing options, it should be under option 5 then option p4 when you run "sudo raspi-config".

thanks - this helped me!