pimoroni/st7789-python #4: Raspberrypi4-st7789 not show

owlhor, 2020-10-22:
I'm trying to use Raspberrypi4 with st7789. scrolling_text.py works. But There're nothing happen in the display. I'm following this for the circuit [ https://i.stack.imgur.com/ouTmF.jpg ].

Gadgetoid, 2020-10-24:
This library was written originally for our Pirate Audio boards and Breakout Garden breakouts, and the examples tend to assume you're using one of those.

For other boards you will need to adjust the examples to use whatever pins you're wiring your breakout to. This is where you would change that in `scrolling-text.py` for example:

https://github.com/pimoroni/st7789-python/blob/fa6195dcb7d12e3969f047ce2b71b2a5b3ee417c/examples/scrolling-text.py#L12-L18

Gadgetoid, 2020-10-24:
The alternative, if you're following the pinout in the linked image, is to change these pins:
[image]

For the ones documented on one of the Pirate Audio breakouts: https://pinout.xyz/pinout/pirate_audio_headphone_amp

SCL|SDA|RES|DC|BLK
----|----|-----|---|---
11  |10  | 22  | 9  | 13

And I'm guessing chip-select on this breakout is tied on to save a pin so you can ignore it.

owlhor, 2020-10-29:
[image]
[image]
[image]

I've tried. It still doesn't work.

Gadgetoid, 2020-10-29:
Well, for starters your backlight is connected to GPIO 13, not GPIO 19 so the code should be updated accordingly (though to be fair, it does look like the backlight is *on* in your picture).

Secondly, perhaps the "res" (Reset, I guess) pin needs to be pulled in some direction for this module to work. I can't seem to find any documentation for it. Might be worth setting GPIO 22 as an output driven HIGH.

Finally, 80MHz might be a *little* optimistic for a breakout at the end of jump wire. Change `80 * 1000 * 1000` to `1 * 1000 * 1000` just to see if *anything* happens, and if it works you can raise the SPI speed until it breaks and drop back a little.

owlhor, 2020-10-29:
Still doesn't work. I tried  `GPIO.output(22, GPIO.HIGH)` , connect `BLK` to `GPIO 19 (pin 35)` and change SPI speed.

Gadgetoid, 2020-10-29:
I don't there's anything else I can suggest- you should contact the seller of the breakout for further support.

closed 2020-10-29 by Gadgetoid