pimoroni/st7789-python #8: Dimming / backlight control on/off

kokosowy, 2020-12-30:
Hey!

I'm using pirate audio and it's fun! I have a question, is it possible to allow dimming on this display or turning of backlight completely?

Thanks for letting know!
K

Sebwap, 2021-11-02:
No return on this since nearly one year ?
I'm asking myself the same question for the new display hat mini which is driven by the same library.
For me, you can shutdown backlight by pulling down the GPIO used for backlight.

Howerver, I don't know if it is possible to do some dimming.

tomtastic, 2022-01-20:
There's a method to set the backlight on/off :

https://github.com/pimoroni/st7789-python/blob/5c16baecb8844b1807da221b40d03dec500b3a9e/library/ST7789/__init__.py#L172

Though I've yet to try it, and there appears no way to adjust the level or dimming of the backlight.

tomtastic, 2022-01-20:
From a quick look at the ST7789 datasheet, it seems there might even be functions to vary brightness that just aren't implemented in the current driver here : https://www.rhydolabz.com/documents/33/ST7789.pdf

`0x51` = Write display brightness
`0x52` = Read display brightness value

[image]

tomtastic, 2022-01-20:
I had a look at RPi.GPIO for the PWM details, it's still software based, but an overlay for the device tree to allow hardware PWM (via syses) looks straightforward enough and would at least avoid burning user cycles and would hopefully equate to less flicker during garbage collection, etc.

ThecaTTony, 2022-06-25:
This is a good example of dimming:

https://github.com/pimoroni/pirate-audio/blob/master/examples/backlight-pwm.py

Wikinaut, 2023-03-03:
I use the PWM method but noticed annoying flickering - where does it stem from? How can this be overcome?

Wikinaut, 2023-03-03:
Using `pigpio`works perfectly. 

I will publish test script using this method later here.

bjohas, 2023-04-25:
@Wikinaut - would love to see the example!

Wikinaut, 2023-04-26:
@bjohas 
Hi, here is the updated code using `pigpio`. Make sure to run pigpiod with `-t 0` option, otherwise you cannot play audio.

https://github.com/Wikinaut/pinetradio/blob/main/backlight-pwm.py

bjohas, 2023-04-26:
Thank you so much!