pimoroni/st7789-python #9: DAC audio being lost when using the screen

duracell80, 2021-01-29:
On the pirate audio board, is there anything in the LCD driver here that could be causing the audio to cut out from the DAC when using the screen?

An initial thought I had was the backlight pins, a conflict there or something similar?

duracell80, 2021-01-29:
In image.py in examples the backlight pin is set to 19 possibly causing the sound to drop. Should this be 13?

mo6, 2021-04-23:
I experience this issue also. Sounds works fine, but when after using the image.py script the sounds stops working.

mo6, 2021-04-23:
Commenting the backlight config seems to fix this, though.

mo6, 2021-04-23:
Like this:

`# Create ST7789 LCD display class.
disp = ST7789.ST7789(
    port=0,
    cs=ST7789.BG_SPI_CS_FRONT,  # BG_SPI_CS_BACK or BG_SPI_CS_FRONT
    dc=9,
#    backlight=19,               # 18 for back BG slot, 19 for front BG slot.
    spi_speed_hz=80 * 1000 * 1000,
    offset_left=40 if display_type == "round" else 0
)`

laust0rm, 2021-05-06:
I used this library, but 1. the display is only showing white, probably because I used a 480x320 screen and the resolution is wrong, so not actually a big deal, 2. it doesn't turn off. is there a function of the display object that resets it? also, i haven't connected the reset pin, however I don't know if that has something to do with this problem.

laust0rm, 2021-05-06:
Okay, I'm a little dumb and after 10 minutes of digging into the __init__() of the lib I found out that not only can you specify the reset pin, you can also set the resolution, so sorry I was posting before thinkig. :/

josephernest, 2021-12-14:
Worked for me with `backlight=13`.

thirdr, 2025-01-07:
Closed as resolved.

closed 2025-01-07 by thirdr