pimoroni/st7789-python #20: using ST7789 library for Mini Display Hat... How do I control the onboard LED?

Nwhitten, 2021-11-14:
Hi

when using the ST7789 library (not the displayhatmini library) is there a way to control the onboard LED??

thanks

Neil

Sebwap, 2021-11-14:
Yes probably, by replicating the code included in display hat mini library!

Wikinaut, 2023-05-01:
see https://github.com/Wikinaut/pinetradio/blob/main/ledtest.py

Wikinaut, 2023-05-01:
You do not need the library. Just control the `ACT` LED on pin 47.

See for example lines in https://github.com/Wikinaut/pinetradio/blob/main/pinetradio.py (look for `ACT`, which stands for activity LED.).

PI system use of the ACT-LED must be disabled. You can use this service https://github.com/Wikinaut/pinetradio/blob/main/disable-led.service and run `sudo systemctl enable disable-led` once.