pimoroni/st7789-python #21: Python vs Python3 apt install

lucasra1, 2021-11-24:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):
Raspbian 32-Bit Light - Raspberry Pi Zero W 2

> Distributor ID:	Raspbian
> Description:	Raspbian GNU/Linux 11 (bullseye)
> Release:	11
> Codename:	bullseye

> Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux

- Error message you are receiving, including any Python exception traces: 

> package python-pip is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
>   python3-pip
> 
> Package python-rpi.gpio is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> Package python-numpy is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> Package python-pil is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'python-rpi.gpio' has no installation candidate
> E: Package 'python-pip' has no installation candidate
> E: Package 'python-pil' has no installation candidate
> E: Package 'python-numpy' has no installation candidate

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
`sudo apt install python-rpi.gpio python-spidev python-pip python-pil python-numpy`

what worked for me was:
`sudo apt install python3-rpi.gpio python3-spidev python3-pip python3-pil python3-numpy`

I dont know if this is just with newer releases of raspbian, but maybe this should be mentioned in the installation guide

mi-der, 2022-01-02:
Having the same issue, but it does not pertain to this library or this repository.  It appears that almost all python packages in the new Raspberry Pi OS package manager are broken.

Gadgetoid, 2022-01-04:
Looks like I updated the instructions as of 5c16baecb8844b1807da221b40d03dec500b3a9e but these changes haven't yet made it to a release.

closed 2022-02-15 by lucasra1