pimoroni/st7789-python #22: st7789/__init__.py  suggestion for two small changes

PaulskPt, 2021-12-06:
No problem encountered. Just a suggestion for two small changes. It is too few to arrange a PR. That is why I do my suggestion in this way.
When I was looking into this st7789 module  my eye 'fell' onto the following:
in st7789/__init__.py, inside the class ST7789, function __init__(),  Current situation: a) function reset() is called even when a reset pin has not been provided. Suggestion: move the call to self.reset() inside the if statement block so that a call to self.reset() only occurs when a reset pin is provided; b) suggestion to change the 'if rst is not None:' into 'if self._rst is not None:' as is done inside the reset() function while in line 129 is stated: 'self._rst = rst' . See the composed image. 

[image]

Gadgetoid, 2022-01-04:
These make sense! Thank you.

A PR would still be accepted if you'd like to make the contribution, otherwise I'll try and batch this change in before the next release.

Gadgetoid, 2024-11-12:
Been a while, but I'm pretty sure these changes made it in! Thank you.

closed 2024-11-12 by Gadgetoid