pimoroni/st7789-python #23: displayhatmini-python/examples/pygame-demo.py crash in ST7789/__init__.py line 123

PaulskPt, 2021-12-07:
Platform: RPi400
OS: Raspberry Pi OS version bullseye (32bit)
External devices attached: Pimoroni DisplayHatMini 2.0 on an Adafruit CyberDeck, connected to the 40-pin header of the RPi400.
All relevant OS software and needed modules installed; checked.

/boot/config.txt lines:  
[all]
dtoverlay=spi=on
dtoverlay=displayhatmini

lsmod result: "spi_bcm2835     24576   0"

~/pimoroni/displayhatmini-python/examples/pygame-demo.py crashes 
in ST7789/__init__.py line 123 with [Error 2] FileNotFoundError (see attached screenshot),

[image]

Googled for possible alike solved issues. Not found. Only one case also with FileNotFoundError in connection to spidev.SpiDev() but concerning other hw/sw using linux (just for info: https://forum.pine64.org/showthread.php?tid=6282).
Entered a question on Pimoroni's discord channel #tech-support but no reply (yet).  
That is why I come to bother you here.

PaulskPt, 2021-12-07:
Update: beside the above mentioned issue I now was able to successfully build the Pimoroni/displayhatmini-fbdev-dts/displayhatmini-overlay.dtbo; copy that file to /boot/overlays; do the necessary settings in /boot/config.txt; open F2 console; issue "FRAMEBUFFER=/dev/fb1 fbterm -r 0" and have the console redirected to the Displayhatmini 2.0 But I'm still stuck with that Python script crash detailed above.

Gadgetoid, 2021-12-07:
The Python scripts are intended to talk *directly* to the display and don't work in conjunction with the fbdev driver- which basically takes over the screen entirely.

You'd need to hack on https://github.com/Gadgetoid/displayhatmini-fbdev-dts/blob/main/demo.py which squirts its data out to the framebuffer device. Though probably not while also running fbterm since that would also vy for control of the screen.

Everything gets messy when you don't have GPU compositing handling all this stuff gracefully 😢

PaulskPt, 2021-12-07:
Hi Philip, thank you for your rapid reply. The addition about fbdev was only to indicate that I was progressing in getting things on the device. Reading your reply do I interprete well that - when I want to use the Python scripts - I have to comment-out the dtparam lines in /boot/config.txt b4 ?

PaulskPt, 2021-12-07:
I just commented out the lines:
dtoverlay=spi0-1cs,cs0_pin=7,cs1_spidev=disabled
dtoverlay=displayhatmini
Then went to ~/pimoroni/displayhatmini-python/examples and ran ./pygame-demo.py.
Now no crash; backlight on; LED on right side of display off. Black screen.
Text in the terminal window: 
"pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Display HAT Mini: Pygame Demo"
and after Ctrl+C
"Exiting!..."

E-man-dev, 2022-05-15:
I am also getting an error where
`disp = ST7789.ST7789(
     width=240,
     height=240,
     rotation=90,
     port=0,
     cs=ST7789.BG_SPI_CS_FRONT,
     dc=9,
     backlight=19,
     spi_speed_hz=80 * 1000 * 1000,
     offset_left=0,
     offset_top=0
)`
results in a traceback to
'File "/usr/local/lib/python3.9/dist-packages/ST7789/__init__.py", line 123, in __init__'
where
`self._spi = spider.SpiDev(port, cs)`
results in
'FileNotFoundError: [Errno 2] No such file or directory'.
I am using a Pirate Audio Mol Speaker hat, which uses the 1.3" display. I am running the latest 32-bit Raspberry Pi OS on a Raspberry Pi 3B+. The default username is not being used.

helgibbons, 2022-05-16:
@E-man-dev Have you got SPI turned on in `raspi-config` ?

E-man-dev, 2022-05-17:
@helgibbons I double checked and SPI is turned on in `raspi-config`

E-man-dev, 2022-07-25:
A fresh OS image fixed the problem for me.

treatmesubj, 2022-08-09:
Thanks @helgibbons I had SPI turned off.
$ sudo raspi-config
3. Interface Options
I4 SPI
Enable