pimoroni/st7789-python #26: Package doesn't declare dependencies

tilusnet, 2022-05-10:
This package should declare all library dependencies. I don't think it does.
Eg: `spidev`, `numpy`, etc. won't get installed if someone installs this library.

cpietsch, 2022-10-17:
In https://github.com/pimoroni/pirate-audio there is an entry for that. Missing here though.
```
sudo apt-get update
sudo apt-get install python-rpi.gpio python-spidev python-pip python-pil python-numpy
```

johnbchron, 2022-11-02:
Not sure if this is necessary but the explicitly python3 version of that is
```
sudo apt-get install python3-rpi.gpio python3-spidev python3-pip python3-pil python3-numpy
```

darac, 2024-06-16:
Is there a reason that these requirements are "hidden" in the documentation, rather than being declared in a machine readable manner?

It is _usually_ expected that, the following would work:
```
❯ python -m venv .venv
❯ . .venv/bin/activate
❯ pip install st7789
Collecting st7789
  Downloading st7789-1.0.0-py3-none-any.whl.metadata (5.6 kB)
Collecting gpiod (from st7789)
  Downloading gpiod-2.1.3.tar.gz (53 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 53.1/53.1 kB 776.5 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting gpiodevice>=0.0.4 (from st7789)
  Downloading gpiodevice-0.0.5-py3-none-any.whl.metadata (4.0 kB)
Downloading st7789-1.0.0-py3-none-any.whl (10 kB)
Downloading gpiodevice-0.0.5-py3-none-any.whl (9.2 kB)
Building wheels for collected packages: gpiod
  Building wheel for gpiod (pyproject.toml) ... done
  Created wheel for gpiod: filename=gpiod-2.1.3-cp311-cp311-linux_x86_64.whl size=93452 sha256=c63c1a4b3af6628e7b160ff7c50e8afb785b56ed06d3a8baaa9abe1f9c7e22c2
  Stored in directory: /home/ipa/darac/.cache/pip/wheels/81/1c/ec/7b8a80a113b96dd990c7f4e41af5d2e2cb1e99a67c2a4bb110
Successfully built gpiod
Installing collected packages: gpiod, gpiodevice, st7789
Successfully installed gpiod-2.1.3 gpiodevice-0.0.5 st7789-1.0.0
❯ python
Python 3.11.9 (main, Apr 10 2024, 13:16:36) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import st7789
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/tmp/test/.venv/lib/python3.11/site-packages/st7789/__init__.py", line 26, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'
>>> 
```

Gadgetoid, 2024-06-16:
> Is there a reason that these requirements are "hidden" in the documentation, rather than being declared in a machine readable manner?

historically yes

now? Not so much!

small stuff-up on my part

PR #32 (2024-06-17): pyproject.toml: Add additional dependencies.
Should fix #26
closed 2024-06-17 by Gadgetoid