pimoroni/st7789-python #31: ST7789V Variant?

nebhead, 2024-06-14:
Any plans to support the ST7789**V** variant?

Gadgetoid, 2024-06-17:
I'd say the chances are somewhere above zero if I got my hands on one. Unfortunately most of the work on this library is driven by product requirements, and I don't have any such requirements, nor much free time at the moment!

Pull requests would be graciously accepted.

mander1000, 2024-08-09:
I added the variant: https://github.com/mander1000/st7789-python

Unfortunately I also switched from `gpiod` to `RPi.GPIO` (as RPi.GPIO uses MIT and not LGPL) and added a function to deactivate and (re-)activate the display (as I wanted to switch the display on/off later). What do you want to merge? ;)

Gadgetoid, 2024-08-09:
As weird and painful as it is, you should be using the Linux standard gpiod. Anything else is either wrong, broken, soon to be broken, slow, or just uses gpiod under the hood anyway.

MIT vs LGPL is largely irrelevant when the *entire OS* is GPL anyway 😆