pimoroni/st7789-python #34: Dual displays no longer functional on newest library version

mitgobla, 2024-07-25:
Previously, you could run two SPI displays (I was able to run two 240x240 round displays) at the same time, so long that the CS and BL pins were different.

For reference, this was discussed on the forums and this is where I was able to implement it from: [https://forums.pimoroni.com/t/two-1-3-spi-colour-lcd-240x240-on-one-pi/16737/8](https://forums.pimoroni.com/t/two-1-3-spi-colour-lcd-240x240-on-one-pi/16737/8)

Now, when attempting to run the `gif.py` or `image.py` example with a minor addition to initialize a copy of `disp` as `disp2` with the CS and BL pins changed, I get the following error message:
```python
Traceback (most recent call last):
  File "/home/pi/Downloads/st7789-python/examples/gif.py", line 57, in <module>
    disp2 = st7789.ST7789(
  File "/usr/local/lib/python3.9/dist-packages/st7789/__init__.py", line 154, in __init__
    self._dc = gpiodevice.get_pin(dc, "st7789-dc", OUTL)
  File "/usr/local/lib/python3.9/dist-packages/gpiodevice/__init__.py", line 163, in get_pin
    lines = chip.request_lines(consumer=f"{consumer}-{label}", config={line_offset: settings})
  File "/home/pi/.local/lib/python3.9/site-packages/gpiod/chip.py", line 315, in request_lines
    req_internal = self._chip.request_lines(line_cfg, consumer, event_buffer_size)
OSError: [Errno 16] Device or resource busy
```

It appears that the latest version no longer releases the DC pin after use and therefore the second display cannot be initialized.

Steps:

1. Create display such as those in the examples with cs as BG_SPI_CS_BACK and backlight as 18
2. Create second display with cs as BG_SPI_CS_FRONT and backlight as 19
3. Run the code. Error occurs when second display is initialized.

- Platform: Raspberry Pi 4B with Raspian
- Python version: 3.9.2

mitgobla, 2024-07-25:
I've been able to confirm, returning to version 0.0.4 I am able to have both displays use the same DC port. Version 1.0.0 and latest 1.0.1 return the error mentioned above.

Gadgetoid, 2024-11-11:
This is a fundamental issue with the GPIO Character Device ABI on Linux, and how it (arguably correctly) handles mutually exclusive ownership of pins.

That said, it should be possible to work around it by providing the same "pin" object to each instance of the display driver.

Unfortunately it currently doesn't support that, since it does: https://github.com/pimoroni/st7789-python/blob/f7b386ec58a7a5e29a16fc10f26306375bc179c8/st7789/__init__.py#L154

It should be reasonable to modify the library to check if `dc` here is a tuple (lines/offset) and *not* initialise it. Then init becomes something like (heavily abridged):

```python
dc = gpiodevice.get(dc, "st7789-dc", OUTL)

display1 = st7789(0, 0, dc)
display2 = st7789(0, 1, dc)
```

commit 0b3cba8e (2024-11-11): Support shared pins.

Allow the same pin instance to be passed into multiple driver instances.

Fixes #34.
closed 2025-01-14 by Gadgetoid