pimoroni/st7789-python #36: RPI 5 "Mmap of GPIO registers failed"

mbdev22, 2024-11-23:
- Platform/operating system :  
RPI5 8GB
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
aarch64 (64-Bit)

- Python version (run `python -version` or `python3 -version`): Python 3.7.3

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "examples/scrolling-text.py", line 46, in <module>
    offset_left=40 if display_type == "round" else 0
  File "/usr/local/lib/python3.7/dist-packages/ST7789/__init__.py", line 134, in __init__
    GPIO.setup(dc, GPIO.OUT)
RuntimeError: Mmap of GPIO registers failed

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
Have a 1.3" Pimoroni LCD SPI display on a breakout garden board in the front slot. Running the command 'python3 examples/scrolling-text.py "Hi" square' from the example folder gives the Mmap error message above. On previous versions below the RPI5 it was working correctly, displaying the provided text input.

Gadgetoid, 2024-11-24:
Looks like an out of date library, that error is from RPi.GPIO which we don't use anymore!

Try:

`pip install --upgrade st7789`

mbdev22, 2024-11-25:
Thank you for the response, unfortunately still no luck there:

pip install --upgrade st7789
   Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
   Requirement already up-to-date: st7789 in /usr/local/lib/python2.7/dist-packages (0.0.4)

pip uninstall st7789
Uninstalling ST7789-0.0.4:
  Would remove:
    /usr/local/lib/python2.7/dist-packages/ST7789-0.0.4.dist-info/*
    /usr/local/lib/python2.7/dist-packages/ST7789/*
Proceed (y/n)? y
  Successfully uninstalled ST7789-0.0.4

pip install st7789
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting st7789
Installing collected packages: st7789
Successfully installed st7789-0.0.4

Tried upgrading and reinstalling, is 0.0.4 the current version?
Could that be related to the RPI5 as this was working on my old Pi 4 without issues?

Gadgetoid, 2024-11-25:
> Tried upgrading and reinstalling, is 0.0.4 the current version?

0.0.4 is ancient, there's been a 1.0.0 and 1.0.1 release since.

Oooh you're using Python 2.7. You should be using Python 3! 2.7 is very, very EOL.

mbdev22, 2024-11-26:
Thanks a lot for pointing that out, after upgrading the issue is fixed. 
Now I am just getting a "AttributeError: module 'gpiod' has no attribute 'Chip'" error, 
but that seems to be just a local package or module problem.

mbdev22, 2024-11-28:
Thanks again, got it working by installing inside a virtuial environment

closed 2024-11-28 by mbdev22